NotPunchnox/rkllama #1: Errors on trying to start + documentation inconsistency

Nurb4000, 2025-01-12:
I have noticed some inconsistencies in the directions, which may be due to language translation issues, but also getting a hard error when trying to run.

1 - While a git clone uses lower case, in several spots in the starter documentation it shows upper case. When ran, it looks for an uppercase folder in the users home directory. 

2 - Early on in documentation it says to place models in the ./models directory, but then later says to put then in the root of the ( capitalized ) folder.   Just to be safe, i copied it to both locations. 

I renamed the folder that was cloned to all caps to get past the first error where its looking for the files in the capitalized folder name. However the below error is received on run of "rkllama", even if i just pass in the help parameter...  I am starting to try to read thru the code, but i don't know french, so its difficult for me.

I am running a fresh install of the same version of Ubuntu as you have stated, however it is not an OPI, but a FriendlyArm board, as it has more ram and storage options, but still has an RK3588.

PS: thanks for the work you are doing. 

~~~~~~~~~~~~~~~~~~~~~~~~~~~

File "/home/user/RKLLAMA/client.py", line 134
    print(f"\n\n{GREEN}Tokens par seconde{RESET}: {final_json["usage"]["tokens_per_second"]}")
                                                               ^^^^^
SyntaxError: f-string: unmatched '['

~~~~~~~~~~~~~~~~~~~~~~~~~~~

NotPunchnox, 2025-01-13:
Hello, thank you for your comments. I'm going to translate the code comments into English to make them more understandable, and I'll try to improve the explanations/screenshots that are out of date.

I'll take a closer look at the errors you mentioned.

I'll try to fix all that in the next few days, if not tonight.

Thanks for your help and see you soon (I'll keep you posted as soon as I can).

NotPunchnox, 2025-01-14:
Hi, tomorrow everything should be worked out, I will send a post here to detail how to fix your mistake and explain the changes made 👍

NotPunchnox, 2025-01-15:
Hello,

I have made the necessary improvements and fixed the issues you pointed out.

### Python Error

Regarding the Python error, it was caused by the version of Python being used. Additionally, there was another issue related to a switch (switch case...) in the `client.py` code. This feature only works correctly from Python 3.10 onwards. I have made the necessary adjustments, and you should no longer encounter errors on that front.

### Model Installation

As for the model installation, you were right to point out that my documentation was not clear enough. I have rewritten the instructions in the `README.md` file and added a new command, `rkllama pull`, which allows you to directly download models from Hugging Face. If you prefer to install the models manually, you can simply place them in the `~/RKLLAMA/models/` folder.

### Software Update

To update the software, you just need to run the `setup.sh` script. If you encounter any issues, try running the script from a directory other than `~/RKLLAMA`.

### Final Step

Finally, once Rkllama is installed, you can delete the original folder, as it will be cloned into the `~/RKLLAMA` directory.

### Code and Comments Translation

I also took care of translating the entire code and comments into English to make it more understandable!

---

I hope I could help, and if you face any other issues, don't hesitate to let me know.

Have a great day!

Nurb4000, 2025-01-15:
Great, thank you, and ill give it a try tomorrow

Nurb4000, 2025-01-16:
Had a few other oddities come up. Relied on your code to install minconda and before the requirements were being installed it didn't initialize so instead of installing it complained about installing in the base OS without the override parameter  ( which id expect if it wasn't initialized yet )

So went and initialized conda by hand and got thru the install just fine but got an error on run.

I suspect the device is confused at this point from all the gyrations, and instead of trying to clean it up by hand, im just going to blow away the OS this weekend and start over.  Install  and initialize miniconda by hand and be sure its working before i start your installer, since i see you do check first.

Once its all done ill report back.

NotPunchnox, 2025-01-17:
Have you tried running the program without virtualizing the python environment?

I'll see if I can do a pull request without conda

Nurb4000, 2025-01-17:
Closer :)

I had a couple of minutes to try today unexpectedly, ill have to continue tomorrow. I realize this isn't real detailed, sort of in a hurry.

Setup conda. Created an environment for python 3.9. Named it RK. Installed dependencies via conda, however the version of flask wasn't available via conda so i need to see what is up with that tomorrow too.  I removed the version, which may cause me other issues, but i just wanted to see how far i could get.

Ran the installer, still got the 'bla bla dont install via root' which is odd since i was IN a venv to start with.  Not had time to look at the non-conda installer file yet.  

After installer was done, no RKLLAMA folder, seems it had created it in the root home folder instead of my user's home folder.

Since im pressed for time today, I took the easy route and just logged in as root, created a new env, bla bla  and did the install after installing the requirements. Of course I got the same warning about installing into the system as root . Its like its dropping out of the environment during the install, but it does not seem to be. Its odd. Will take some reading on my part.

Trying to start the server i get the below error, which might be due to the flask version, or some dependencies not getting installed due to the 'install to system' error.

This weekend ill look at the flask version issue in conda, and peek at the installer for the pip paramaters. I will also see if i can get a hold of the OPI tonight and start fresh with that instead tomorrow.

Actual wording of the 'warning' is below too, which may be the root ( no pun intended ) of my problem now.  My first thought is ill just add the parameter so it doesn't complain and worry about why later. Right now, even it was not in a conda env, it would be dedicated to this anyway since its all testing right now.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
New error:

rk) root@CM3588-Plus:~/src/rkllama-Without-miniconda# rkllama serve
Traceback (most recent call last):
  File "/root/RKLLAMA/server.py", line 231, in <module>
    main()
  File "/root/RKLLAMA/server.py", line 212, in main
    print_color("Error argument not found: --target_platform")
TypeError: print_color() missing 1 required positional argument: 'color'
(rk) root@CM3588-Plus:~/src/rkllama-Without-miniconda#
   

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Pip error:

WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager, possibly rendering your system unusable.It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv. Use the --root-user-action option if you know what you are doing and want to suppress this warning.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Nurb4000, 2025-01-17:
Ok i had a meeting cancelled and i was able to grab the OPI5.  

I did not notice a link to the exact OS you were using the first time, and just the version, but saw it now, and i went and got the most current for the OPI there so it matches 100% to your setup.  

Just for refrence it was this file ->  ubuntu-24.10-preinstalled-server-arm64-orangepi-5.img.xz

Ran the installer, it worked this time with no root complaints and it did create the directory in the user folder, not root, so something must be off between distributions ( which is odd at this level . its just python and conda and simple shell scripts.. but whatever, its better :) ) However, got the same error i got on the other machine when starting the server with "rkllama serve" :

~~~~~~~~~~~~~~~~~~~~~~~~
Traceback (most recent call last):
  File "/home/ubuntu/RKLLAMA/server.py", line 231, in <module>
    main()
  File "/home/ubuntu/RKLLAMA/server.py", line 212, in main
    print_color("Error argument not found: --target_platform")
TypeError: print_color() missing 1 required positional argument: 'color'
~~~~~~~~~~~~~~~~~~~~~~~~

NotPunchnox, 2025-01-17:
Good evening, I've found the problem and hope that it will finally work, I've just made a commit that should resolve the error: [commit 65](https://github.com/NotPunchnox/rkllama/commit/bf2afae5a2019b37edff3de09cb391d6d629c59a)

Thank you very much for your detailed reports, it helps me a lot to fix all the possible and imaginable errors.

Nurb4000, 2025-01-18:
Got home earlier than expected so gave it a try ( still on the 16GB OPI 5 )..    Yay it worked :)

That last change got things working fine. 

I used one of the OS images from the link you had, and that also got rid of the 'pip as root' and all the other issues with the server.py fix.  However.... it seems those images don't have the driver baked in, so instead of messing with that myself i just went and grabbed one from Pelochus github which has the drivers baked in. They are based on Armbian so i had to do the 'install stuff via root' stuff instead of the conda, but in my case, its going to be dedicated so it wont hurt anything. 

I first tried with the regular tree, then I did switch over to the version without conda to try it too, and updated serer.py with the fix there too, and it seems to do what it needs to do  ( manually installing requirements ) but i did notice when the serer starts it does say "Starting the environment with Miniconda3" but i assume that is just left over stuff in the file since its not really doing that ( i didn't go look, wasn't a big deal to me but might confuse others ) Also saw a message right afterward "Error argument not found: --target_platform" but it didn't seem to hurt anything so again, i think its just a bogus error.  ( see below for the actual text )

Thank you so much :)   Looking forward to your conversion system too. That is still a pain.

Next week ill see if i can get that 32G board working.  Bigger models. yay!

Below is the actual startup message using the non-conda tree, and again, the 'target platform' error effected nothing. It worked fine, just thought you would like to see it.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

$ rkllama serve
Starting the environment with Miniconda3.
Error argument not found: --target_platform
Start the API at http://localhost:8080
 * Serving Flask app 'server'
 * Debug mode: off
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:8080
 * Running on http://10.0.1.36:8080
Press CTRL+C to quit
127.0.0.1 - - [17/Jan/2025 20:59:28] "GET / HTTP/1.1" 200 -
127.0.0.1 - - [17/Jan/2025 20:59:28] "GET /current_model HTTP/1.1" 404 -
I rkllm: rkllm-runtime version: 1.1.4, rknpu driver version: 0.9.8, platform: RK3588

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

NotPunchnox, 2025-01-18:
Perfect, I'm glad it finally worked. As for the error in the “without miniconda” branch, I've just corrected it in the latest commit.

I'll be able to start coding the template conversion software.

I think I'll close this topic if that's okay with you. Feel free to open a new one if necessary!

Nurb4000, 2025-01-18:
Ya, we are good to close.   I didn't want to until you saw the note about the stray conda message since while it wasn't hurting, it would confuse people.   Now its off to play with the API and try to get that 32 G board running for more headroom.    

Have a great weekend :)

NotPunchnox, 2025-01-20:
Hi, sorry it took me a while to reply, I've been pretty busy lately.
You were right not to close the commit before I saw the message.

Thanks again for your contribution, have a great week and maybe see you soon!

closed 2025-01-20 by NotPunchnox

NotPunchnox, 2025-02-16:
Hello, I've made a small update. The `without-miniconda` branch will be deleted soon. Now, if you want to use RKLLAMA without Miniconda, simply run the `setup.sh` file with the `--no-conda` argument. All files like `client.sh` and `server.sh` will automatically use this argument.

```bash
./setup.sh --no-conda
```

Have a nice day!

Nurb4000, 2025-02-16:
ya. having the other branch for long term would be confusing to most.