NotPunchnox/rkllama #4: Support for DeepSeek R1

svesnav, 2025-01-27:
I conveted DeepSeek R11.5B in RKLLM, But it doesn't work correctly and writes very strange reflections. I have tested this model on my computer at ollama and it responds adequately there. 

Running on Orange Pi 5 Plus 16GB:

[image] 

Running the same model on PC:

[image]

Nurb4000, 2025-01-27:
I wonder if its the 'thoughts' stuff it feeds back at first that is causing the issue.    ( may not be, just a thought , and no pun intended )

NotPunchnox, 2025-01-28:
It is probably due to the "special tokens," which might be different from those predefined in the rkllama Python code. If you want to try resolving the issue on your side while I look into it, you just need to modify this line:  
```py
PROMPT_TEXT_PREFIX = "<|im_start|>system You are a helpful assistant. <|im_end|> <|im_start|>user"
PROMPT_TEXT_POSTFIX = "<|im_end|><|im_start|>assistant"
```
In this file (https://github.com/NotPunchnox/rkllama/blob/main/src/classes.py)  
Then reinstall the software using:  
```bash
sudo ./setup.sh
```  
The changes will be applied automatically.

However, I did a test earlier with [DeepSeek-Distill-Qwen-1.5B](https://huggingface.co/jbower/DeepSeek-R1-Distill-Qwen-1.5B-rk3588-1.1.2) on rkllama, and I had no issues with the special tokens. However, the generated response would not stop.

I will take a closer look at this tomorrow and keep you updated on the progress. I will make a commit to fix the issue as soon as possible.

svesnav, 2025-01-28:
Thank you for your reply, this is not an urgent (at least for me) problem, so you can solve it as needed. I will be watching the commits with impatience!

NotPunchnox, 2025-02-03:
The error you are encountering comes from special tokens. I will try to create chat templates based on all compatible models or add the AutoTokenizer function from HuggingFace to resolve the error and provide support for all models. I will keep you updated on the progress.

NotPunchnox, 2025-02-03:
Ok it's working fine ( tests were done on DeepSeek:7b and Llama:3b ), I'm working on the [“Support All Models”](https://github.com/NotPunchnox/rkllama/tree/Support-All-models) branch, but before merging this branch with the [“main”](https://github.com/NotPunchnox/rkllama/) branch, I need to make a lot of changes in the code to implement the Modelfile system like Ollama does.

I won't have time to finish this tonight, but it'll happen soon enough!

svesnav, 2025-02-03:
This is very good news! Thank you so much for that!

Nurb4000, 2025-02-03:
cool

NotPunchnox, 2025-02-04:
Good evening,

I have great news to share: DeepSeek is now officially supported by RKLLAMA!

I’ve implemented a major update that requires running a few commands to reorganize the files in the RKLLAMA/models/ directory, but the process is quite fast.

- [Main documentation](https://github.com/NotPunchnox/rkllama/blob/main/README.md)
- [New file organization documentation](https://github.com/NotPunchnox/rkllama/blob/main/documentation/Guide/en/Rebuild-arch.md)

I’ve also updated the documentation and created a new guide to help you transition from the old architecture to the new one. This step is only necessary if you want to avoid re-downloading models already present in ~/RKLLAMA/models/.

If you run the `rkllama pull` command, the transition to the new architecture will take place automatically for the model downloaded. 

You'll find more details in the documentations above.

Feel free to ask any questions if needed.

Wishing you a great evening!

svesnav, 2025-02-04:
This is very cool news, I will definitely check how it works tomorrow! Thank you very much for solving the problem!

svesnav, 2025-02-05:
Please add transformers to the requirements.txt file, as it is missing during the installation of rkllama.

NotPunchnox, 2025-02-05:
It's done 👍

closed 2025-02-05 by svesnav