NotPunchnox/rkllama #7: Request docker support

kylin7226, 2025-02-08:
I hope it can run in a docker environment
thx！

Nurb4000, 2025-02-12:
While im not a fan personally of Docker ( i prefer 100% isolation ), it would open this up to a lot more people.

ichlaffterlalu, 2025-02-15:
> Yes, that's a good idea. I just created a Dockerfile and updated the documentation on a branch named [Docker](https://github.com/NotPunchnox/rkllama/tree/Rkllama-Docker). Unfortunately, I won't be able to try this out until Saturday. If everything works perfectly during my tests, I will merge the Docker branch with the main branch.

Hello, @NotPunchnox, I am currently trying to help you test your app and Dockerfile. I made my own pull request to your `Rkllama-Docker` branch, because I made some changes along the way.

NotPunchnox, 2025-02-15:
Hello @ichlaffterlalu,

I've merged your pull request, and everything is working perfectly. This should resolve the issue. I’ve also added you as a contributor in the [README.md](https://github.com/NotPunchnox/rkllama/blob/main/README.md) file because your help saved me a tremendous amount of time.

Once again, a huge thank you!

@kylin7226, I'll let you test it on your end. If everything works fine for you, I'll go ahead and close the issue.

ichlaffterlalu, 2025-02-16:
Is there any date when that Docker version released to main, and maybe to upload your own public Docker image?

closed 2025-02-16 by NotPunchnox

NotPunchnox, 2025-02-16:
Hello,  

I've finished setting everything up. Everything has been merged into the `main` branch. I had to fix a few issues, such as the processor not being detected in `server.sh`...  

Anyway, everything is good now! Here are the commands to use the RKLLama Docker image:  

### Pull the Docker image:  
```bash
docker pull ghcr.io/notpunchnox/rkllama:main
```

### Run the server:  
```bash
docker run --rm -it -p 8080:8080 ghcr.io/notpunchnox/rkllama:main
```

Have a great day! Hope this helps.