NotPunchnox/rkllama #13: CPU not detected??

Grime-Magnolia, 2025-02-27:
on commit 6c88fbe13cee6065b957f476bb2c4bfbf5c228a1
using rock 5b+ radxa
using reborn os arm - based on arch linux

here is the log
```log
tygo@Rock5bp ~/RKLLAMA (main)> rkllama serve
Starting the environment with Miniconda3.
CPU_MODEL: 
Erreur : CPU model not found !
tygo@Rock5bp ~/RKLLAMA (main)> lscpu
Architecture:           aarch64
  CPU op-mode(s):       32-bit, 64-bit
  Byte Order:           Little Endian
CPU(s):                 8
  On-line CPU(s) list:  0-7
Vendor ID:              ARM
  Model name:           Cortex-A55
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 4
    Socket(s):          1
    Stepping:           r2p0
    CPU(s) scaling MHz: 100%
    CPU max MHz:        1800,0000
    CPU min MHz:        408,0000
    BogoMIPS:           48,00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomi
                        cs fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
  Model name:           Cortex-A76
    Model:              0
    Thread(s) per core: 1
    Core(s) per socket: 2
    Socket(s):          2
    Stepping:           r4p0
    CPU(s) scaling MHz: 100%
    CPU max MHz:        2400,0000
    CPU min MHz:        408,0000
    BogoMIPS:           48,00
    Flags:              fp asimd evtstrm aes pmull sha1 sha2 crc32 atomi
                        cs fphp asimdhp cpuid asimdrdm lrcpc dcpop asimd
                        dp
Caches (sum of all):    
  L1d:                  384 KiB (8 instances)
  L1i:                  384 KiB (8 instances)
  L2:                   2,5 MiB (8 instances)
  L3:                   3 MiB (1 instance)
Vulnerabilities:        
  Itlb multihit:        Not affected
  L1tf:                 Not affected
  Mds:                  Not affected
  Meltdown:             Not affected
  Spec store bypass:    Mitigation; Speculative Store Bypass disabled vi
                        a prctl
  Spectre v1:           Mitigation; __user pointer sanitization
  Spectre v2:           Vulnerable: Unprivileged eBPF enabled
  Srbds:                Not affected
  Tsx async abort:      Not affected
```

TomJacobsUK, 2025-02-27:
Had same issue with orange-pi 5+

Added a prompt on server startup, probably could be better handled in a config file

https://github.com/TomJacobsUK/rkllama/commit/5f995778b4c1a765e79b272ce75ea9950b28b9fd

NotPunchnox, 2025-02-28:
@TomJacobsUK  If you’d like, you can submit a pull request so that your menu will be included in the official version of rkllama, and you’ll appear as a contributor.  

Otherwise, I can add your work while mentioning you.  

It’s entirely up to you! Thanks again for the help you’ve provided!

Nurb4000, 2025-02-28:
for what its worth, im seeing it tonight too, fresh install on a friendlyelec CM3588 ( NAS )

TomJacobsUK, 2025-03-01:
Added pull request for my workaround.

NotPunchnox, 2025-03-01:
Super thanks for your contribution, I've accepted your pull request

closed 2025-03-02 by Grime-Magnolia