NotPunchnox/rkllama #15: Minor issue: CPU_MODEL environment variable doesn't work at all due to this line of code.

ichlaffterlalu, 2025-03-02:
Seems like the CPU_MODEL variable that should be using the environment variable value, is always overridden by this variable instantiation.
https://github.com/NotPunchnox/rkllama/blame/cc787496e3c2819f5a3d11e349b14f0614c92213/server.sh#L8

This caused running RKLLAMA via Docker in a device that triggered #13 issue (has no CPU model name stated in `/proc/cpuinfo`), stuck in infinite loop after changes merged from #14.

The solution is simple: remove that `CPU_MODEL=""` line.

NotPunchnox, 2025-03-04:
Alright, thanks for the information. I'll take a look at it this evening and make a commit accordingly.

Thanks, and have a great day!

TomJacobsUK, 2025-03-07:
Added a pull request to fix not using the ENV and infinite loop due too non-interactive.

PR #16 (2025-03-07): CPU selection
This pull request includes updates to the server configuration and startup script to improve flexibility and default settings. The most important changes setting default values in the script, and enhancing CPU model detection.

Updates to server configuration:

* [`rkllama.ini`](diffhunk://#diff-e9756ab56f97af210d85b8b670b1bf36527a162ebb9cc63cecae2ee7c0bc423dL2-R2): Changed the server port from 8081 to 8080 due to other references being 8080

Enhancements to startup script:

* [`server.sh`](diffhunk://#diff-14d1967cdbd3938fb362411b147b4e44ab53b6a45a2c11a6bdd0c0d3c1658e54L57-R73): [... cut]
closed 2025-03-07 by NotPunchnox