NotPunchnox/rkllama #18: Proposal: Unified Configuration System for RKLLAMA

TomJacobsUK, 2025-03-08:
I was going to address this as part of some of my ollama-api #17  updates, but think it's best to break it into smaller chunk and get some thoughts before proceeding.

# Proposal: Unified Configuration System for RKLLAMA

## Current Challenges

Our RKLLAMA codebase faces several configuration-related issues:

1. **Hardcoded Absolute Paths**: Many paths assume installation at `~/RKLLAMA/`, limiting deployment options
2. **Inconsistent Configuration**: Settings scattered across Python code and shell scripts
3. **Limited Flexibility**: Difficult to deploy in containers or non-standard environments
4. **No Clear Configuration Precedence**: Unclear which settings take priority

## Proposed Solution

A unified configuration system integrating both Python code and shell scripts:

### Core Components

1. **Configuration Hierarchy**:
   ```
   Command-line Args → Environment Variables → Project INI → User INI → System INI → Defaults
   ```

2. **Relative Path Resolution**:
   - Application root directory as the base for all relative paths
   - Support for absolute paths when explicitly needed

3. **Shell/Python Integration**:
   - Auto-generated shell-compatible config from INI files
   - Common environment variables shared between shells and Python code

4. **Centralized Configuration**:
   - One configuration module that all components access
   - Clear validation and error reporting for missing/invalid settings

### Implementation Structure

```
RKLLAMA/
├── config/
│   ├── default.ini        # Default configuration
│   └── config.env         # Auto-generated shell config
├── config.py              # Configuration management module
├── server.py
└── server.sh              # Sources config.env
```

### Configuration Process Flow

1. On startup, Python code determines the application root
2. Configuration loads from all sources in priority order
3. Generates `config.env` for shell script consumption
4. All paths resolve relative to application root unless absolute

### Shell Script Integration

Shell scripts will:
1. Determine their location to find application root
2. Source the auto-generated `config.env` file
3. Allow command-line arguments to override defaults
4. Pass configuration to Python scripts when launching them

## Migration Path

1. Create configuration module with current defaults
2. Replace hardcoded paths one component at a time
3. Update shell scripts to use the new configuration system
4. Add documentation for configuration options

## Benefits

- **Portable Installation**: Deploy anywhere without code modifications
- **Container-Friendly**: Easy deployment in Docker environments
- **Simplified Development**: Test with different configurations easily
- **Clear Precedence**: Obvious which setting takes priority
- **Single Source of Truth**: All settings defined in one place
- **Self-Documentation**: Configuration file documents available options

## Next Steps

1. Create initial `config.py` module and default configuration
2. Update paths in core components (library loading, model paths)
3. Implement shell script integration
4. Document configuration options and precedence

Would you like me to proceed with implementation or provide more specific details on any aspect of this proposal?

NotPunchnox, 2025-03-09:
@TomJacobsUK It sounds very clear to me, I really like the proposal. It’s true that it would make deployments and configuration much easier.

If you want to start the implementation, I have no issue with that. I’ll follow the progress, and if you need any help, don’t hesitate to ask here.

In the meantime, I’ll start looking into implementing the embedding models.

TomJacobsUK, 2025-03-14:
Added pull request that addresses fixes the hard coded paths and added config management.

[Documentation here](//github.com/NotPunchnox/rkllama/blob/8b93a1517bac7d341a67953bfd71a7a9733f851b/documentation/configuration.md)

One nice feature is can now move models to home directory `~/models` and use them in multiple rkllama instances/branches