NotPunchnox/rkllama #20: No need to connect to huggingface once.

fellow99, 2025-03-17:
My RKLLAMA would be deployed in an offline env.
So it could not connect to huggingface.
I checked the code. And It connect to huggingface to get tokenizer from the specified repo.
Could the Modelfile sets a local cache path to store files?
`
tokenizer = AutoTokenizer.from_pretrained(model_name, cache_dir=......)
`

TomJacobsUK, 2025-03-17:
Makes sense think we could possibly pass path to the model `pretrained_model_name_or_path` and improve whats download/stored from huggingface initially.

NotPunchnox, 2025-03-24:
I'll see about adding that to the Modelfile.

Nurb4000, 2025-03-24:
I can see that being good, not everyone may have their device connected to the outside, security reasons.

NotPunchnox, 2025-04-12:
I’ve added the ability to import a custom tokenizer, enabling offline deployment.
**Important**: You must use a full absolute path and avoid using `~`. The server runs with administrative permissions by default due to the `fix-frequence.sh` script, so an incomplete path will cause it to fail.

I’ve also included safeguards to prevent the program from crashing if the path is incorrect. If the specified path isn’t found, it will automatically fall back to the Hugging Face tokenizer. Any issues will be logged in the server logs.

I hope this will be helpful!

[image]

[image]

fellow99, 2025-04-18:
I reviewed new codes.
Only `/generate` calls `load_tokenizer` in `process.py`, but `/api/generate` or `/apt/chat` does not.
In most cases, we use ollama API directly, or OpenAI API.

Iyozhikov, 2026-01-18:
Here is my change to `src/rkllama/api/server_utils.py`
```
class EndpointHandler:
    """Base class for endpoint handlers with common functionality"""

    @staticmethod
    def prepare_prompt(
        model_name, messages, system="", tools=None, enable_thinking=False
    ):
        """Prepare prompt with proper system handling"""
        custom_tokenizer_value = (
            get_property_modelfile(
                model_name, "TOKENIZER", rkllama.config.get_path("models")
            )
            .replace('"', "")
            .replace("'", "")
        )
        logging.debug(f"Custom Tokenizer is: {custom_tokenizer_value} ")
        # Check if the custom tokenizer path exists
        if os.path.exists(custom_tokenizer_value):
            try:
                # Attempt to load the custom tokenizer
                tokenizer = AutoTokenizer.from_pretrained(
                    custom_tokenizer_value, trust_remote_code=True
                )
                logging.info(f"Loaded custom tokenizer from {custom_tokenizer_value}")
            except Exception as e:
                # Warn user and prepare to fallback
                logging.info(
                    f"Warning: Could not load tokenizer from {custom_tokenizer_value}. \nError: {str(e)}. Falling back to default tokenizer."
                )
        else:
            # Warn user if path is invalid
            logging.info(
                f"Warning: Tokenizer path {custom_tokenizer_value} does not exist.\nFalling back to default tokenizer."
            )

        if not tokenizer:
            # Get model specific tokenizer from Huggin Face specified in Modelfile
            model_in_hf = (
                get_property_modelfile(
                    model_name, "HUGGINGFACE_PATH", rkllama.config.get_path("models")
                )
                .replace('"', "")
                .replace("'", "")
            )

            # Get the tokenizer configured for the model
            tokenizer = AutoTokenizer.from_pretrained(
                model_in_hf, trust_remote_code=True
            )
        supports_system_role = (
            "raise_exception('System role not supported')"
            not in tokenizer.chat_template
        )

        if system and supports_system_role:
            prompt_messages = [{"role": "system", "content": system}] + messages
        else:
            prompt_messages = messages

        prompt_tokens = tokenizer.apply_chat_template(
            prompt_messages,
            tools=tools,
            tokenize=True,
            add_generation_prompt=True,
            enable_thinking=enable_thinking,
        )

        return tokenizer, prompt_tokens, len(prompt_tokens)
```

Model file TOKENIZER parameter:

```
FROM="Qwen3-8B-rk3588-w8a8-opt-0-hybrid-ratio-0.0.rkllm"

TOKENIZER="/root/.cache/huggingface/hub/models--dulimov--Qwen3-8B-rk3588-1.2.1-unsloth-16k/snapshots/2db47ba60c7c14a4b440c4c27d821c69e02ea927/"

HUGGINGFACE_PATH="dulimov/Qwen3-8B-rk3588-1.2.1-unsloth-16k"

```
Contents of model directory (directory will be created automatically by rkllama -> huggunface calls at normal conditions - before you'll set TOKENIZER option and apply change from above, or you can download those files manually or with huggingface cli)

```
ls /root/.cache/huggingface/hub/models--dulimov--Qwen3-8B-rk3588-1.2.1-unsloth-16k/snapshots/2db47ba60c7c14a4b440c4c27d821c69e02ea927
added_tokens.json    merges.txt               tokenizer_config.json  vocab.json
chat_template.jinja  special_tokens_map.json  tokenizer.json
```

Sytem log entries (Custom Tokenizer) - no connections to Huggungface:

```
Jan 18 14:34:25 station-m3 rkllama-server[6029]: I rkllm: Enabled cpus: [4, 5, 6, 7]
Jan 18 14:34:25 station-m3 rkllama-server[6029]: I rkllm: Enabled cpus num: 4
Jan 18 14:34:26 station-m3 rkllama-server[5613]: 2026-01-18 14:34:26,039 - rkllama.worker - INFO - Worker for model Qwen3:8b created and running...
Jan 18 14:34:26 station-m3 rkllama-server[5613]: 2026-01-18 14:34:26,044 - rkllama.server - DEBUG - Model Qwen3:8b loaded successfully
Jan 18 14:34:26 station-m3 rkllama-server[5613]: 2026-01-18 14:34:26,044 - root - DEBUG - Custom Tokenizer is: /root/.cache/huggingface/hub/models--dulimov--Qwen3-8B-rk3588-1.2.1-unsloth-16k/snapshots/2db47ba60c7c14a4b440c4c27d821c69e02ea927/
Jan 18 14:34:26 station-m3 rkllama-server[5613]: 2026-01-18 14:34:26,582 - root - INFO - Loaded custom tokenizer from /root/.cache/huggingface/hub/models--dulimov--Qwen3-8B-rk3588-1.2.1-unsloth-16k/snapshots/2db47ba60c7c14a4b440c4c27d821c69e02ea927/
Jan 18 14:34:26 station-m3 rkllama-server[5613]: 2026-01-18 14:34:26,652 - rkllama.server - DEBUG - Releasing lock in chat_ollama
Jan 18 14:34:26 station-m3 rkllama-server[6029]: 2026-01-18 14:34:26,652 - rkllama.worker - INFO - Running inference for model Qwen3:8b...
```