NotPunchnox/rkllama #22: unable to load the rkllm file

bhodgens, 2025-03-20:
After installing and pulling an rkllm on a radxa rock3b board, I'm unable to load the model: 
```
rock@rock1:~/RKLLAMA$ rkllama load DeepSeek-R1-Distill-Llama-8B-rk3588-w8a8-opt-1-hybrid-ratio-0.0.rkllm
Unknown command: load.
```

It would appear that the client.py doesn't have any functions for loading the model at all. `config.env` has RKLLAMA_MODEL_DEFAULT but this appears to get overwritten on `serve`. 

Running from this commit: 

```
rock@rock1:~/RKLLAMA$ git show --summary
commit 9c3e524bf9fada6cdba54ee382ea518813c99d86 (HEAD -> main, origin/main, origin/HEAD)
Merge: 77f5c55 5580e8f
Author: NotPunchnox <74469788+NotPunchnox@users.noreply.github.com>
Date:   Sat Mar 15 15:57:15 2025 +0100

    Merge pull request #19 from TomJacobsUK/config

    Flexible configuration system for RKLLAMA

 create mode 100644 config.py
 create mode 100644 config/default.ini
 create mode 100644 config_schema.py
 create mode 100644 documentation/configuration.md
 delete mode 100644 rkllama.ini
```

NotPunchnox, 2025-03-24:
Hello, it's true that the `load` command is no longer available; I will make an update to fix this.

However, if you want it to work, you simply need to run the following command (the `run` command launches the model in the background anyway):

```bash
rkllama run DeepSeek-R1-Distill-Llama-8B-rk3588-w8a8-opt-1-hybrid-ratio-0.0.rkllm
```

NotPunchnox, 2025-03-24:
**The error has been fixed!**
commit: fc15b31

> Demo:
> [image]`

closed 2025-03-26 by NotPunchnox