NotPunchnox/rkllama #23: Questions regarding the RKLLM supported quantization types for RK3588

limcheekin, 2025-03-21:
Hi there,

I think it is a good idea to make RKLLM models a litter bit more user friendly to newcomer (I'm one of them) by creating a reference table to let the user to have some general idea which quantization type they should use, resources needed and what quality they will get in return. I point this out from the perspective of an user of ollama and GGUF models.

I get the following answer from NotebookLM, is it correct?
> The following quantization types are supported by the RK3588 platform, specifically in the context of RKLLM according to source:
> - **w8a8:** This quantization type signifies 8-bit asymmetric quantization for both the weights and the activation values in the neural network model. This means that both the parameters (weights) of the model and the intermediate computations (activations) are represented using 8-bit fixed-point numbers with an asymmetric quantization scheme.
> - **w8a8_g128:** This quantization type represents 8-bit grouped quantization for both weights and activations, with a group size of 128. In grouped quantization, the weights and activations are divided into groups of 128 elements, and quantization parameters (like scale and zero-point) are shared within each group. This can offer a better trade-off between model size reduction, inference speed, and accuracy compared to per-tensor or per-channel quantization.
> - **w8a8_g256:** Similar to w8a8_g128, this type indicates 8-bit grouped quantization for both weights and activations, but with a larger group size of 256. Increasing the group size can potentially lead to further model size reduction and faster inference, but might also result in a slight decrease in accuracy depending on the model and data.
> - **w8a8_g512:** This is another 8-bit grouped quantization scheme for both weights and activations, with an even larger group size of 512. As the group size increases further, the potential benefits in terms of size and speed might continue, but the risk of accuracy degradation could also increase.
> 
> It's important to note from source that for these grouped quantization types ("w8a8_g128", "w8a8_g256", "w8a8_g512"), the group size must divide the output dimension of the linear layer; otherwise, the quantization process will fail.
> Additionally, our conversation history mentions that the RKNN SDK supports linear asymmetric INT8 quantization [our conversation history]. The "w8a8" type aligns with this, where 'w8' signifies 8-bit weights and 'a8' signifies 8-bit activations, both using an asymmetric quantization scheme as indicated by source. The grouped versions extend this by applying quantization at a group level within the tensors.
> The availability of these specific "w8a8" grouped quantization types ("g128", "g256", "g512") for RK3588 is explicitly mentioned in the context of the RKLLM SDK, which is designed for large language models. The standard RKNN Toolkit 2 also supports "w8a8", which likely forms the basis for these grouped quantization variants in the LLM SDK for the RK3588 platform.

From the explanation above, it look like `w8a8` will generate the best quality response and required most resources, is that correct?

Still have `optimizations` and `hybrid_rates`, let's take the models I converted as example:
https://huggingface.co/limcheekin/Light-R1-14B-DS-rk3588-1.1.4/tree/main

Is the optimized version `opt-1` such as `Light-R1-14B-DS-rk3588-w8a8-opt-1-hybrid-ratio-0.5.rkllm` generally better performance (quality of response) than the unoptimized version `opt-0` such as `Light-R1-14B-DS-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm`?

How is the  `hybrid-ratio-0.0`, `hybrid-ratio-0.5` and `hybrid-ratio-1.0` works in term of response quality?

ichlaffterlalu, 2025-04-02:
Hello, @limcheekin,

Sorry that I might be too late to answer.
You can read more about the RKLLM quantization types, compared to Ollama ones, in this document: https://github.com/NotPunchnox/rkllama/blob/main/documentation/api/quantization_mapping.md

About the optimization flag (`opt-1` or `opt-0`), yes, the optimized ones have generally better performance than the unoptimized one.

About "hybrid ratio", sadly I also didn't understand what that means... I've asked Google and LLMs, both are not giving satisfiable answer, often calling the term "too broad, depends on the context".

Hope that helps.

ichlaffterlalu, 2025-04-02:
Oh, I think I can include GitHub Copilot's answer about the "hybrid ratio" thing here:

> In the context of custom Large Language Models (LLMs), the **"hybrid ratio"** typically refers to a parameter or metric used to balance or combine different types of computation, data representation, or model components. While the exact meaning can vary depending on the implementation, here are some common interpretations:
> 
> 1. **Quantization Precision Balancing**:  
>    In mixed-precision quantization, the hybrid ratio might define the proportion of weights or activations that are stored or computed at higher precision (e.g., FP16) versus lower precision (e.g., INT8). For example, in a "w8a16" model, the weights are quantized to 8 bits, and activations are at 16 bits. A hybrid ratio could control the trade-off between precision and efficiency.
> 
> 2. **Model Compression Techniques**:  
>    When combining different compression strategies (e.g., pruning, quantization, or distillation), the hybrid ratio might indicate the proportion of the model or layers subjected to each technique. For example, part of the model might use sparse weights, while another part uses quantized weights.
> 
> 3. **Compute Resource Allocation**:  
>    In distributed or hybrid inference setups (e.g., CPU + GPU or GPU + NPU), the hybrid ratio might define the percentage of computation performed on each type of hardware. This can optimize performance by leveraging the strengths of multiple hardware types.
> 
> 4. **Training vs. Inference Optimization**:  
>    For custom models, the hybrid ratio could also refer to the balance between resources allocated for training versus inference. For example, during fine-tuning, some layers might remain frozen (pre-trained) while others are updated, and the hybrid ratio could define this balance.
> 
> 5. **Token Mixing in Multi-Modal Models**:  
>    In multi-modal LLMs (e.g., combining text and image inputs), the hybrid ratio might represent the proportion of tokens or embeddings from each modality used in the model's input or attention layers.
> 
> In summary, the **hybrid ratio** is a flexible term that generally describes a balance or proportion between two or more components in a custom LLM model. Its specific meaning depends on the context, such as quantization, resource allocation, or multi-modal processing.

closed 2025-04-27 by NotPunchnox