NotPunchnox/rkllama #27: No response from the model

chedim, 2025-03-29:
Freshly installed rkllama, no responses from the model listed in your examples that I pulled with rkllama:

```
FROM: TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
HuggingFace Path: punchnox/Tinnyllama-1.1B-rk3588-rkllm-1.1.4
```

[image]

Please advise?

gonesurfing, 2025-03-30:
Same issue, Rock-5c with Armbian.

malakraft, 2025-03-30:
Hello, got the same issue the other day, I noticed that my kernel didn't had NPU driver. Maybe try to see if drivers are install. 

Using that command: `sudo cat /sys/kernel/debug/rknpu/version`
Should output something like: `RKNPU driver: v0.9.7`

I initially install ubuntu-24.10 from Joshua-Riek (which didn't had NPU drivers). Had to pass to ubuntu-24.04 for RKLLAMA to work.

Hope it help

Nurb4000, 2025-03-30:
> Same issue, Rock-5c with Armbian.

Last time i looked at stock Armbian it didn't have the RK drivers baked in.  Had to use a 'remix'  ( https://github.com/Pelochus/armbian-build-rknpu-updates )

gonesurfing, 2025-03-31:
The Armbian kernel I'm on has the NPU driver included.
`Linux rock-5c 6.1.99-vendor-rk35xx #1 SMP Fri Feb  7 15:19:27 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux`

`sudo cat /sys/kernel/debug/rknpu/version`
`RKNPU driver: v0.9.8`

from dmesg:
`[drm] Initialized rknpu 0.9.8 20240828 for fdab0000.npu on minor 1`

chedim, 2025-03-31:
Mine is rockchip-ubuntu 24.10

```
root@agatha:/home/chedim# uname -a
Linux agatha 6.1.0-1025-rockchip #25-Ubuntu SMP Mon Aug 26 23:01:14 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
root@agatha:/home/chedim# cat /sys/kernel/debug/rknpu/version
RKNPU driver: v0.9.7
root@agatha:/home/chedim# 
```

chedim, 2025-04-05:
😿 anything I can do to help with this?

RikudouGoku, 2025-04-09:
> The Armbian kernel I'm on has the NPU driver included. `Linux rock-5c 6.1.99-vendor-rk35xx #1 SMP Fri Feb 7 15:19:27 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux`
> 
> `sudo cat /sys/kernel/debug/rknpu/version` `RKNPU driver: v0.9.8`

I also have the same exact RKNPU driver version although im on the Orange pi 5 plus.
Linux orangepi5-plus 6.1.99-vendor-rk35xx #1 SMP Wed Mar 19 11:16:36 UTC 2025 aarch64 GNU/Linux

I am using the docker installation though but exactly the same issue as OP has.

> from dmesg: `[drm] Initialized rknpu 0.9.8 20240828 for fdab0000.npu on minor 1`

How do I see this?

RikudouGoku, 2025-04-09:
> 😿 anything I can do to help with this?
Here are the log for my container after just starting it.
rkllama-1  | Starting the environment with Miniconda3.
rkllama-1  | Processor: rk3588
rkllama-1  | RK3588 frequencies optimized for NPU inferencing
rkllama-1  | Setting the frequency for the rk3588 platform...
rkllama-1  | Initializing model mappings...
rkllama-1  | Start the API at http://localhost:8080
rkllama-1  |  * Serving Flask app 'server'
rkllama-1  |  * Debug mode: off
rkllama-1  | 2025-04-09 18:55:41,504 - werkzeug - INFO - WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
rkllama-1  |  * Running on all addresses (0.0.0.0)
rkllama-1  |  * Running on http://127.0.0.1:8080
rkllama-1  |  * Running on http://192.168.144.2:8080
rkllama-1  | 2025-04-09 18:55:41,505 - werkzeug - INFO - Press CTRL+C to quit

RikudouGoku, 2025-04-09:
@chedim @gonesurfing I fixed it!

[image]

I am not sure which of these 2 changes i made did this but here is the updated docker compose. I added the Devices section and changed the image.
(Got this from ChatGPT ironically.)
```
version: "3.8"
services:
  rkllama:
    container_name: rkllama
    image: ghcr.io/notpunchnox/rkllama:main
    privileged: true
    ports:
      - 11434:11434 # Ollama-compatible endpoint
    volumes:
      - /docker/rkllama/models:/root/RKLLAMA/models
    devices:
      - /dev/dri/renderD129:/dev/dri/renderD129 # NPU device
    restart: unless-stopped
```

As for why I changed the port, i did that due to ChatGPT:

> The port has changed from the default 8080:8080 to 11434:11434 because Ollama-compatible API on rkllama (the image you’re using) uses port 11434 for communication.
> 
> Here’s the reasoning behind it:
> 
> Default Ports:
> 
> 8080 is often used for web interfaces or services in many applications, but it can be used for any service.
> 
> RKLLAMA and Ollama Compatibility:
> 
> rkllama implements Ollama's API and listens on port 11434 by default for compatibility with Ollama clients.
> 
> This allows you to interact with the server using the standard Ollama API endpoints (/api/generate, /api/chat, etc.).
> 
> Why the Change:
> 
> The 11434 port is specifically chosen to avoid conflicts with other applications that may already be using port 8080.
> 
> It’s the port expected by most Ollama-compatible clients like Open WebUI.

This is the old compose I used that did not work for reference.
```
name: rkllm-server
services:
  rkllama:
    volumes:
      - /docker/rkllm-server/models:/root/RKLLAMA/models
    privileged: true
    ports:
      - 8091:8080
    image: thanhtantran/rkllama:arm64
```

chedim, 2025-04-12:
@RikudouGoku I really don't believe your issue is the same as mine. I was not using docker in my tests and the fix that you proposed worked for you because you mapped the NPU device into the container which does not make sense if one doesn't run rkllama in a conainer in the first place. Also, you're using a different board and I think that my problem is coming from using an untested board and not from docker configuration. I hope the developer would be able to resolve the issue and I am here for them to help with anything I can do.

chedim, 2025-04-12:
Found how to enable debugging mode. Here is the server output: 

```
root@agatha:~/RKLLAMA# DEBUG_MODE=1 rkllama serve
Starting the environment with Miniconda3.
Processor: rk3588
Debug mode enabled
2025-04-12 01:28:34,453 - rkllama.config - INFO - Current RKLLAMA Configuration:
2025-04-12 01:28:34,453 - rkllama.config - INFO - [server]
2025-04-12 01:28:34,454 - rkllama.config - INFO -   port = 8080
2025-04-12 01:28:34,454 - rkllama.config - INFO -   host = 0.0.0.0
2025-04-12 01:28:34,454 - rkllama.config - INFO -   debug = True
2025-04-12 01:28:34,454 - rkllama.config - INFO - [paths]
2025-04-12 01:28:34,454 - rkllama.config - INFO -   models = models
2025-04-12 01:28:34,454 - rkllama.config - INFO -   logs = logs
2025-04-12 01:28:34,454 - rkllama.config - INFO -   data = data
2025-04-12 01:28:34,454 - rkllama.config - INFO -   src = src
2025-04-12 01:28:34,454 - rkllama.config - INFO -   lib = lib
2025-04-12 01:28:34,454 - rkllama.config - INFO -   temp = temp
2025-04-12 01:28:34,454 - rkllama.config - INFO - [model]
2025-04-12 01:28:34,454 - rkllama.config - INFO -   default = 
2025-04-12 01:28:34,454 - rkllama.config - INFO - [platform]
2025-04-12 01:28:34,454 - rkllama.config - INFO -   processor = rk3588
Setting the frequency for the rk3588 platform...
NPU available frequencies:
300000000 400000000 500000000 600000000 700000000 800000000 900000000 1000000000
Fix NPU max frequency:
1000000000
CPU available frequencies:
408000 600000 816000 1008000 1200000 1416000 1608000 1800000 
408000 600000 816000 1008000 1200000 1416000 1608000 1800000 2016000 2208000 2304000 
408000 600000 816000 1008000 1200000 1416000 1608000 1800000 2016000 2208000 2352000 
Fix CPU max frequency:
1800000
2304000
2352000
GPU available frequencies:
1000000000 900000000 800000000 700000000 600000000 500000000 400000000 300000000
Fix GPU max frequency:
1000000000
DDR available frequencies:
528000000 1068000000 1560000000 2112000000
Fix DDR max frequency:
2112000000
Initializing model mappings...
Start the API at http://localhost:8080
 * Serving Flask app 'server'
 * Debug mode: on
2025-04-12 01:28:34,520 - werkzeug - INFO - WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:8080
 * Running on http://192.168.220.11:8080
2025-04-12 01:28:34,521 - werkzeug - INFO - Press CTRL+C to quit
2025-04-12 01:28:34,521 - werkzeug - INFO -  * Restarting with stat
2025-04-12 01:28:39,003 - rkllama.config - DEBUG - Generated shell configuration: /root/RKLLAMA/config/config.env
2025-04-12 01:28:39,004 - rkllama.config - DEBUG - Generated shell configuration: /root/RKLLAMA/config/config.env
2025-04-12 01:28:39,005 - rkllama.config - DEBUG - Generated shell configuration: /root/RKLLAMA/config/config.env
Debug mode enabled
2025-04-12 01:28:39,005 - rkllama.config - INFO - Current RKLLAMA Configuration:
2025-04-12 01:28:39,005 - rkllama.config - INFO - [server]
2025-04-12 01:28:39,005 - rkllama.config - INFO -   port = 8080
2025-04-12 01:28:39,005 - rkllama.config - INFO -   host = 0.0.0.0
2025-04-12 01:28:39,005 - rkllama.config - INFO -   debug = True
2025-04-12 01:28:39,005 - rkllama.config - INFO - [paths]
2025-04-12 01:28:39,005 - rkllama.config - INFO -   models = models
2025-04-12 01:28:39,005 - rkllama.config - INFO -   logs = logs
2025-04-12 01:28:39,005 - rkllama.config - INFO -   data = data
2025-04-12 01:28:39,005 - rkllama.config - INFO -   src = src
2025-04-12 01:28:39,005 - rkllama.config - INFO -   lib = lib
2025-04-12 01:28:39,005 - rkllama.config - INFO -   temp = temp
[... 25 lines cut ...]
Initializing model mappings...
2025-04-12 01:28:39,071 - rkllama.model_utils - DEBUG - Mapped model: Qwen2.5-1.5B -> qwen2.5:1b
2025-04-12 01:28:39,071 - rkllama.model_utils - DEBUG - Mapped model: TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5 -> tinyllama-chat:1.1b
2025-04-12 01:28:39,071 - rkllama.model_utils - DEBUG - Mapped model: deepseek-r1-1.5B-rkllm1.1.4 -> deepseek:1.5b
Start the API at http://localhost:8080
2025-04-12 01:28:39,076 - werkzeug - WARNING -  * Debugger is active!
2025-04-12 01:28:39,077 - werkzeug - INFO -  * Debugger PIN: ----------
2025-04-12 01:30:12,829 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:12] "GET / HTTP/1.1" 200 -
2025-04-12 01:30:12,834 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:12] "GET /models HTTP/1.1" 200 -
2025-04-12 01:30:28,978 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:28] "GET / HTTP/1.1" 200 -
2025-04-12 01:30:32,513 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:32] "POST /generate HTTP/1.1" 400 -
2025-04-12 01:30:38,829 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:38] "GET / HTTP/1.1" 200 -
2025-04-12 01:30:38,834 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:30:38] "GET /models HTTP/1.1" 200 -
2025-04-12 01:31:06,238 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:31:06] "GET / HTTP/1.1" 200 -
FROM: TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
HuggingFace Path: punchnox/Tinnyllama-1.1B-rk3588-rkllm-1.1.4
I rkllm: rkllm-runtime version: 1.1.4, rknpu driver version: 0.9.7, platform: RK3588

2025-04-12 01:31:07,612 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:31:07] "POST /load_model HTTP/1.1" 200 -
2025-04-12 01:31:14,425 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:31:14] "GET / HTTP/1.1" 200 -
2025-04-12 01:31:16,190 - urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): huggingface.co:443
2025-04-12 01:31:16,592 - urllib3.connectionpool - DEBUG - https://huggingface.co:443 "HEAD /punchnox/Tinnyllama-1.1B-rk3588-rkllm-1.1.4/resolve/main/tokenizer_config.json HTTP/1.1" 200 0

2025-04-12 01:31:16,727 - werkzeug - INFO - 127.0.0.1 - - [12/Apr/2025 01:31:16] "POST /generate HTTP/1.1" 200 -
```

NotPunchnox, 2025-04-12:
> [@chedim](https://github.com/chedim) [@gonesurfing](https://github.com/gonesurfing) I fixed it!
> 
> [image]
> 
> I am not sure which of these 2 changes i made did this but here is the updated docker compose. I added the Devices section and changed the image. (Got this from ChatGPT ironically.)
> 
> ```
> version: "3.8"
> services:
>   rkllama:
>     container_name: rkllama
>     image: ghcr.io/notpunchnox/rkllama:main
>     privileged: true
>     ports:
>       - 8091:8080 
>     volumes:
>       - /docker/rkllama/models:/root/RKLLAMA/models
>     devices:
>       - /dev/dri/renderD129:/dev/dri/renderD129 # NPU device
>     restart: unless-stopped
> ```
> 
> This is the old compose I used that did not work for reference.
> 
> ```
> name: rkllm-server
> services:
>   rkllama:
>     volumes:
>       - /docker/rkllm-server/models:/root/RKLLAMA/models
>     privileged: true
>     ports:
>       - 8091:8080
>     image: thanhtantran/rkllama:arm64
> ```

Are you getting these results with the Tinnyllama-1.1B-rk3588-rkllm-1.1.4 model ?

NotPunchnox, 2025-04-12:
I'm going to add a feature to add a custom/imported tokenizer without needing an internet connection, and I'll look into this bug coming from my model, it's true that after the version 0.0.2 update it doesn't work anymore and that's on my side too,

I'll keep you updated soon, in any case all the other templates work without problem so I recommend you use other models for now, thanks for your understanding!

RikudouGoku, 2025-04-12:
> [@RikudouGoku](https://github.com/RikudouGoku) I really don't believe your issue is the same as mine. I was not using docker in my tests and the fix that you proposed worked for you because you [...]

True, you arent using the docker version and I am but if you are having problems with the regular version I suggest you try the docker one. As the logs and symptoms are the exact same as I had before.

> Are you getting these results with the Tinnyllama-1.1B-rk3588-rkllm-1.1.4 model ?

Yes. But I also tried running other models and they work as well. These:
https://huggingface.co/c01zaut/Phi-3-mini-128k-instruct-RK3588-1.1.4/blob/main/Phi-3-mini-128k-instruct-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
https://huggingface.co/c01zaut/Llama-3.2-3B-Instruct-rk3588-1.1.4/blob/main/Llama-3.2-3B-Instruct-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm

cxtal, 2025-12-04:
I am running in docker but I either do not get any response from Tinyllama or the container crashes when running Qwen.