NotPunchnox/rkllama #30: Embeddings API

alexeyvolkoff, 2025-04-13:
What is actually the issue with Embeddings API? Is it just not yet implemented or some RKLMM framework limitation?

alexeyvolkoff, 2025-04-15:
Great! In a meanwhile, is SentenceTransformers/all-MiniLM-L6-v2 on host CPU a feasible workaround for RAG? did anyone try it in Open WebUI?

NotPunchnox, 2025-04-16:
I haven't tried it, but it seems like a good idea. Additionally, the issue with an NPU is that it can only run one model at a time, and if you want to do RAG, unloading one model to load another can really slow down inference.

However, the embedding model will inevitably be slower on the CPU than on the NPU, but given the lightweight nature of an embedding model, running it on the CPU seems like the best solution, to be honest.

alexeyvolkoff, 2025-04-18:
By the way I have tried RAG with all-MiniLM-L6-v2 in Open WebUI.  Not bad at all.