NotPunchnox/rkllama #32: Addressing models by tag in API

alexeyvolkoff, 2025-04-15:
This is more feature request than bug
Open Web UI displays models by their tags.

`curl -L http://localhost:8080/api/tags
{"models":[{"details":{"family":"llama","format":"rkllm","parameter_size":"3B","quantization_level":"Q8_0"},"digest":"","model":"qwen2.5-instruct:3b","modified_at":"2025-04-15T13:44:49.744915Z","name":"qwen2.5-instruct:3b","size":3738346748}]}`

[image]

Then chat endpoint fails with 404 - not found

[image]

` rkllama.model_utils - ERROR - Model not found: qwen2.5-instruct:3b
 INFO - 127.0.0.1 - - [15/Apr/2025 19:40:32] "POST /api/chat HTTP/1.1" 404 `

alexeyvolkoff, 2025-04-16:
Fixed and created PR for this issue.

[image]

NotPunchnox, 2025-04-16:
Okay it works, it's great, I'm going to go and see your sweater request right away, thank you!

closed 2025-04-16 by NotPunchnox