NotPunchnox/rkllama #37: How to Use API when it comes to docker?

alexxony, 2025-04-22:
My Machine is Orane pi 5 plus

I followed Docker version.

And I tried pop up API

But..

https://github.com/user-attachments/assets/14061822-1144-4972-919e-259e68c6a127

It did not work.

How can I solve this problem?

mag1art, 2025-04-23:
unfortunately it's still not compatible with Ollama API
but
https://github.com/NotPunchnox/rkllama/blob/main/documentation/api/english.md
here is commands you can use over API

alexxony, 2025-04-23:
> unfortunately it's still not compatible with Ollama API but https://github.com/NotPunchnox/rkllama/blob/main/documentation/api/english.md here is commands you can use over API

In this link, there is no solution..

NotPunchnox, 2025-04-23:
The Docker image version was outdated, so I updated it earlier. I'm currently running tests, but the endpoints for Ollama compatibility are finally working with Docker.

NotPunchnox, 2025-04-23:
However, in the video you sent, the API seems to work fine, at least I don't notice any apparent issues.

alexxony, 2025-04-25:
> However, in the video you sent, the API seems to work fine, at least I don't notice any apparent issues.

I gonna delete and reinstall;  but it will work ?

alexxony, 2025-04-25:
> However, in the video you sent, the API seems to work fine, at least I don't notice any apparent issues.

[image]

[image]

How can I ask and get respond??

NotPunchnox, 2025-04-25:
An API doesn't include a graphical interface. If you want a graphical interface that's compatible with Ollama, I recommend using the extension: Page assist: https://github.com/n4ze3m/page-assist.

You just need to enter the URL where the API is running, and you'll be able to use a graphical interface.

Otherwise, you can only use the CLI app (just the rkllama command in the terminal), which will allow you to download models via Hugging Face, launch models, and more.

closed 2025-04-27 by NotPunchnox

alexxony, 2025-05-08:
> The Docker image version was outdated, so I updated it earlier. I'm currently running tests, but the endpoints for Ollama compatibility are finally working with Docker.

Did you update?