NotPunchnox/rkllama #38: Support rknn-llm 1.2 released.

limcheekin, 2025-04-25:
Hi there,

I think it is great addition if this project support v1.2 of the https://github.com/airockchip/rknn-llm/blob/main/CHANGELOG.md which released 2 weeks ago especially added support for visual multimodal models such as InternVL2, Janus, and Qwen2.5-VL.

Thank you.

NotPunchnox, 2025-04-25:
Oh, that's really great news—thank you for the information! I'm looking forward to trying it out this week. It would be awesome for implementing embeddings, model conversion, and compatibility with multimodal models.

SLenik, 2025-05-13:
I'm looking forward to the next release with support for rknn-llm 1.2.1b1 (capable of running Qwen3 LLM).

hajajmaor, 2025-08-10:
Already suppport Qwen3, but with this commit, not last version:
30a0541ef9c77d274b53227e6a10fc21bc8a47ef

This commit broke the service

limcheekin, 2025-11-12:
I can run the following Qwen3 model using container deployed with the docker compose file:
https://huggingface.co/dulimov/Qwen3-4B-Instruct-2507-rk3588-1.2.2/blob/main/Qwen3-4B-Instruct-2507-rk3588-w8a8-opt-1-hybrid-ratio-0.5.rkllm

Close the issue.

closed 2025-11-12 by limcheekin