NotPunchnox/rkllama #48: where is modelfile

zhjygit, 2025-05-13:
when i run ''rkllama load llama3-8b.rkllm',i fails as follows:Error loading model: 400 - Modelfile not found in 'llama3-8b' directory.

in fact, i have no model file. my llama3-8b.rkllm is converted via rkllm-toolkit, the rkllm-toolkit only convert safetensors models files to rkllm file without Modelfile.

zhjygit, 2025-05-13:
how can i load a local rkllm file and run it.

zhjygit, 2025-05-13:
``` FROM="file.rkllm"

 HUGGINGFACE_PATH="huggingface_repository"

 SYSTEM="Your system prompt"

 TEMPERATURE=1.0

 TOKENIZER="path-to-tokenizer"```

HUGGINGFACE_PATH="huggingface_repository", my internet can not connect to the huggingface, so i can not load the local rkllm.

lingliyi, 2025-08-01:
Hello, I have also encountered this problem
`radxa@rock-5b:~/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5$ rkllama list
Available models:
- TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5
- Qwen2.5-1.5B-Instruct_W8A8_RK3588
radxa@rock-5b:~/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5$ rkllama load TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5
Error loading model: 400 - FROM or HUGGINGFACE_PATH not defined in Modelfile.
radxa@rock-5b:~/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5$ cat ./
Modelfile                                                          TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
radxa@rock-5b:~/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5$ cat ./Modelfile 
FROM /home/radxa/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
TEMPLATE """{{ if .System }}<|system|>
{{ .System }}<|end|>
{{ end }}{{ if .Prompt }}<|user|>
{{ .Prompt }}<|end|>
{{ end }}<|assistant|>
{{ .Response }}<|end|>
"""
PARAMETER stop "<|end|>"
PARAMETER stop "<|eot_id|>"
radxa@rock-5b:~/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5$ pwd
/home/radxa/RKLLAMA/models/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5
`