NotPunchnox/rkllama #60: Can't run model because of "max_context_limit"

54601, 2025-08-02:
I'm using OrangePi plus 5 16G with Ubuntu.

After following the documentation, I successfully installed the program and pulled TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm. 'rkllama list' shows that tinyllama is successfully installed.

However, when try to run the model, I got error:
`Error loading model: 400 - Failed to initialize RKLLM model: -1
Unable to load model tinyllama-chat:1.1b.`

The server prints:
`2025-08-02 13:51:02,391 - werkzeug - INFO - 127.0.0.1 - - [02/Aug/2025 13:51:02] "GET / HTTP/1.1" 200 -
2025-08-02 13:51:02,394 - werkzeug - INFO - 127.0.0.1 - - [02/Aug/2025 13:51:02] "GET /current_model HTTP/1.1" 404 -
FROM: TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
HuggingFace Path: punchnox/Tinnyllama-1.1B-rk3588-rkllm-1.1.4
I rkllm: rkllm-runtime version: 1.2.1, rknpu driver version: 0.9.7, platform: RK3588
I rkllm: loading rkllm model from /home/user1/RKLLAMA/models/tinyllama-chat:1.1b/TinyLlama-1.1B-Chat-v1.0-rk3588-w8a8-opt-0-hybrid-ratio-0.5.rkllm
E rkllm: max_context[16384] must be less than the model's max_context_limit[4096]
2025-08-02 13:51:02,475 - werkzeug - INFO - 127.0.0.1 - - [02/Aug/2025 13:51:02] "POST /load_model HTTP/1.1" 400 -`

I tried another model (https://huggingface.co/VRxiaojie/DeepSeek-R1-Distill-Qwen-7B-RK3588S-RKLLM1.1.4), but that encountered the same issue. I also tried to add "MAX_COTEXT_LIMIT=16384" in ModelFile. Which also didn't work.

Is there any config I did wrong?

54601, 2025-08-02:
> The TinyLlama model is no longer very up to date—in fact, I would even say it's a bit 'dumb'. I recommend trying other models like Qwen3 or LLaMA 3.2 (rkLLM version); there are several available on Hugging Face. Feel free to let me know if the model generates still unsatisfactory or out-of-context responses.

Can you teach me how to check if the model is compatiable? I tried a Deepseek-8b model ([DeepSeek-R1-Distill-Qwen-7B-RK3588S-RKLLM1.1.4](https://huggingface.co/VRxiaojie/DeepSeek-R1-Distill-Qwen-7B-RK3588S-RKLLM1.1.4)) but it doesn't run.

fellow99, 2025-08-04:
I tried the latest version 20250801 with qwen2.5-3B & qwen2.5-coder-3B.
It occured the same error:
`
rkllm: max_context[16384] must be less than the model's max_context_limit[4096]2025-08-04 15:10:22,625 - werkzeug - INFO - 192.168.0.113 - - [04/Aug/2025 15:10:22] "POST /v1/chat/completions HTTP/1.1" 500 
`

mpiotrowski, 2025-08-12:
Any idea how to work around this? I tried just now but unfortunately the same issue

mpiotrowski, 2025-08-21:
> For now the fix for me was editing ~/RKLLAMA/models/{model_name}/Modelfile MAX_NEW_TOKENS=4096 NUM_CTX=4096

This is also the workaround that I ended up using

danielferr85, 2025-09-06:
Hi all, just for clarify... The Modelfile gets created with 16384 because the new library RKLLM (1.2.X) allows context size of this value. If you see that error, is because you are using an old  converted model with library library rkllm 1.1.X that only allows 4096 context size.
Also, during conversion of models with the new library, the user must specify the 16384 context size to take advantage of this improvement, if not, the converted model also stays with 4096 like the old ones. (must people doesn't know this when convert models)

It is more easy that you change the value in the modelfile in all cases:

Let assume that your model was converted to RKLLM with 4096 context:

- You want to work with 4096, you only needs to modify the modelfile and adjust those values to the model loaded by rkllm.
- You want to work with 16328, you can't. You need to convert the original model to RKLLM again with the parameters for 16328 context (more CPU resources = more money). When you have the new model downloaded, nothing need to be done in the modelfile because it came with 16328.

Now, let assume that your model was converted to RKLLM with 16328 context:

- You want to work with 4096, you dont need to modify anything. It works because the modelfile allow context until reach 16328..
- You want to work with 16328, you dont need to modify anything. It works because the modelfile allow context until reach 16328.
- It is better to push the convertion of RKLLM model with context length 16328 with the library version (1.2.X) insted of use the original (and old version) of the library RKLMM (1.1.X) which was limited for 4096 context.

A lot of people that is converting models doesnt know that version 1.2.X of the library allows context of 16328. They need to set this during conversion. If not, they will keep converting with the old smallest context length.

It's kind a way to force to people to start converted the models with their maximun available context size