NotPunchnox/rkllama #62: Please help convert home-3b-v2 to NPU

rasulovk, 2025-08-27:
Hi ,

Could you please advise how to convert [home-3b-v2](https://ollama.com/fixt/home-3b-v2) to version it could be supported by rkllama please

Nurb4000, 2025-11-30:
> Still needed it?

Tho i have not tried for a little while now ( too busy for projects ), some details conversion instructions would be nice.  I have had zero luck following the 'general' directions ( the single command line example ) but all the models, while converted ( i assume ), all failed to load. ( i don't have the exact error handy, but i posted it a while ago )

danielferr85, 2025-11-30:
Ready: https://huggingface.co/danielferr85/home-llama-3.2-3b-w8a8-rk3588 

Steps:

[image]

...

[image]

....

[image]

rasulovk, 2025-12-02:
Thanks a lot, sorry for late responce