NotPunchnox/rkllama #69: AttributeError: module 'config' has no attribute 'validate'

loizres, 2025-09-27:
Since yesterday I am unable to install rkllama using ./setup.sh. The installation of Miniconda itself works fine, but the script fails when trying to run the configuration step.

Output:
`root@jAnI:~/rkllama# ./setup.sh
Miniconda is not installed. Proceeding with installation...
--2025-09-27 13:09:24--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-aarch64.sh
Auflösen des Hostnamens repo.anaconda.com (repo.anaconda.com)… 2606:4700::6810:20f1, 2606:4700::6810:bf9e, 104.16.32.241, ...
Verbindungsaufbau zu repo.anaconda.com (repo.anaconda.com)|2606:4700::6810:20f1|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 155170056 (148M) [application/octet-stream]
Wird in »/tmp/miniconda.sh« gespeichert.

/tmp/miniconda.sh   100%[===================>] 147,98M  11,8MB/s    in 12s

2025-09-27 13:09:49 (12,3 MB/s) - »/tmp/miniconda.sh« gespeichert [155170056/155170056]

PREFIX=/root/miniconda3
Unpacking bootstrapper...
Unpacking payload...

Installing base environment...

Preparing transaction: ...working... done
Executing transaction: ...working... done
installation finished.
Miniconda was successfully installed.
Checking for updates...
Bereits aktuell.
Update check completed successfully!
Installing RKLLAMA to /root/RKLLAMA...
Copying resources to /root/RKLLAMA...
Generating initial configuration...
Traceback (most recent call last):
  File "<string>", line 1, in <module>
    import config; config.validate()
                   ^^^^^^^^^^^^^^^
AttributeError: module 'config' has no attribute 'validate'
`
Environment:

Date the issue appeared: 2025-09-27

OS: Linux (aarch64)

Fresh install, no prior rkllama installation

It seems the config module no longer has a validate() method. Possibly a regression from a recent change?

loizres, 2025-09-27:
For now I could solve the problem with the following commands

`git clone https://github.com/NotPunchnox/rkllama`
`cd rkllama`
`git checkout HEAD~1`

Then continue normal.

danielferr85, 2025-09-28:
Seems that tje last version broke (and maybe deprecated) the classic installation method (setup.sh) and docker deploy. 

I can infer that only the new deployment option is for python package install:

1. Create a virtual env (conda, uv, venv ...)
2. Clone the repo
3. Exexute python -m pip install .
4. Now to start the server the command is rkllama_server and for the client the command is rkllama_client

The problem is that this mode is broken too for dependencies of the location of the RKLLM and RKNN library. If a have time, I will try to fix it. The docker deploynent and the new python package.

It is a shame because I have a new version to upload with other improvements ( like the use of the native rknnlite python library insted of the wrapper to better support), some timeouts for stability, and support for MiniCPM-V-4 multimodal model (now tested support qwen2-vl, qwen2.5-vl and minicpm-v-4..... soon I hope: minicpm-v-4.5)

closed 2025-09-28 by NotPunchnox